ms-iot/virtual-shields-arduino #10: Works with HC-05 and HC-06

MikaelSoderstrom, 2015-05-18:
The readme says that you must use "SparkFun BlueSMiRF Silver". I can verify that it does work with both HC-05 and HC-06 as well.

They were both connected this way:

| Arduino       | Bluetooth module|
| ------------- |:---------------:|
| VCC           | 3.3v            |
| GND           | GND             |
| RX            | TX              |
| TX            | RX              |

Perhaps update the documentation with this information?

JimGaleForce, 2015-05-18:
VERY happy to do so - really glad to add more options. thanks Mikael. Will close this when updated.

JimGaleForce, 2015-05-19:
Mikael: I realize that I have tried one of those in the past and it regularly disconnected after a short time. How long has it worked for you?

MikaelSoderstrom, 2015-05-19:
Haven't tried that long, can test it more soon. I have plans on using it to control an Arduino Robot, so I hope it works well. :)

Will get back when I have tested it again.

MikaelSoderstrom, 2015-05-22:
How short is a "short time"?

I modified the HelloWorld example and changed line 44 to this, so I can see how long it stays connected:

```
screen.printAt(y, String(millis()));
```

I have connected the HC-06 and it's been running for 10 minutes without getting disconnected.

I am going to test it with HC-05 and try that too for 10 minutes.

MikaelSoderstrom, 2015-05-22:
The HC-06 have been running for 20 minutes without getting disconnected. Seems to work without any problems. :)

ooeygui, 2015-08-01:
it looks like this is working for you, so I'm going to close the issue. Reopen if you see problems :-)
Thanks for the report!

closed 2015-08-01 by ooeygui

kumareshmsp221, 2016-03-14:
Hello 
Iam using Microsoft lumia 535 ,HC-05 Bluetooth module and Arduino Mega.I installed windows virtual shield for arduino and I connected to bluetooth when the connection get success after that app get crashed I am unable to get the output please give the solution for that

JimGaleForce, 2016-03-14:
(this is a closed issue - I commented on the other one you opened).