ms-iot/virtual-shields-arduino #15: `VirtualShield::block()` needs comment/correction

zfields, 2015-11-10:
`VirtualShield::block()` makes a peculiar call to `VirtualShield::waitfor()`. If it is used correctly, then the usage should be described in a comment; otherwise it is a bug and should be fixed.

The code in question is pasted below:
```
int block(int id, bool blocking, long timeout = WAITFOR_TIMEOUT, int waitForResultId = -1);
int VirtualShield::block(int id, bool blocking, long timeout, int watchForResultId)
{
    return allowAutoBlocking && blocking ? waitFor(id, timeout, watchForResultId) : id;
}

int waitFor(int id, long timeout = WAITFOR_TIMEOUT, bool asSuccess = true, int resultId = -1);
```

ooeygui, 2015-11-11:
Can you explain fruther what you find peculiar? This reads fine for me.

I think there should be paren around the blocking ternary to be explicit about order of operations.

zfields, 2015-12-09:
`block` calls `waitFor()` with three parameters, which leaves the forth, `resultId` to be defaulted. `block()`'s `waitForResultId` is passed to the `asSuccess` parameter of `waitFor()`, while `resultId` is defaulted to `-1`

ooeygui, 2016-12-07:
Closing issue due to inactivity

closed 2016-12-07 by ooeygui