ms-iot/virtual-shields-arduino #17: Virtual Shield

ImJustSomeGuy, 2015-11-15:
Is it normal for there to be some bug issues?  When I start the Microsoft Virtual Shield app, many times I can not get the ellipsis to do anything.  I've found opening the app with Bluetooth off, turning on Bluetooth, then going back to the app seems to help.  Then, when I connect to the Mate Silver, the message (Hello Virtual Shields....) is slow to come up if at all.  The little microphone in the top right of the screen will be there at time and at times it won't.  The LED will not light often.    Connecting to the Bluetooth also seems finicky and often freezes the app.  I've found hitting Disconnect, then Refresh, then Connect helps.  Lastly, I did the project that uses the 'Windows Remote Arduino Experience' to turn on an external LED with my phone.  Wired exactly the same way as this project, the LED is much brighter than in this project.  I used same output, resistor, everything.  I don't understand why this would be.  Tried it back and forth several times between the two projects.  I am using an Arduino Uno R3 and Sparkfun Bluetooth Mate Silver. I've tried it with both a Lumia 635 & Lumia 928.  Any thoughts/advice would be great.  Thanks

ImJustSomeGuy, 2015-11-16:
Thanks for the information.  As far as debug mode, do you mean in the sketch program using something like 'visual micro', or printing to the serial ports?  Sorry, new to this.  

As far as the LED goes I used the same LED.  In fact, I've gone between the two projects back and forth only changing the program loaded onto the Arduino, and of course from the shield app to the remote app.  I believe pin 8 (which I am using for both) is a digital only output.

turkycat, 2015-11-16:
Arduino debugging is difficult, and yes I've often resorted to printing to serial output or using LEDs to signify when blocks of code are being executed. For the Universal Windows App (for Virtual Shields), you can use Visual Studio to deploy it in debug mode which actually makes debugging much easier.

If your LED isn't bright enough, its OK to switch to a smaller resistor. It might be that the LED you are using is different in some small way. We typically use 330 ohm, but a 220 is just fine as well.

sidwarkd, 2015-12-10:
Closing this issue as an answered question. @ImJustSomeGuy if you have further questions please feel free to reach out.

closed 2015-12-10 by sidwarkd