ms-iot/virtual-shields-arduino #29: Windows Virtual shiled for Arduino

kumareshmsp221, 2016-03-14:
Iam using Microsoft lumia 535 ,HC-05 Bluetooth module and Arduino Mega.I installed windows virtual shield for arduino and I connected to bluetooth when the connection get success after that app get crashed I am unable to get the output please give the solution for that

JimGaleForce, 2016-03-14:
What version of the library (Arduino side) are you using? The one that downloads from the Library Manager is stuck at 1.1.0. You need the 1.2.0 which you can currently get from this repository. We are working with Arduino to update the library through the Library Manager.

kumareshmsp221, 2016-03-14:
Can pls give all the stuff to do work with virtual shield
Arduino version: 1.6.8

JimGaleForce, 2016-03-15:
We will be updating the library and app soon. However, the LATEST code is available here if you are seeing an issue...

The instructions on this page: https://github.com/ms-iot/virtual-shields-arduino 
identify how to download this repository and use as a library for the Arduino IDE.

kumareshmsp221, 2016-03-16:
Hello Sir/Madam

I am currently working with my college project Gps tracking system using UWP ,Arduino and  the  Windows Virtual Shiled for Arduino .In previous query I already mentioned that I am unable to use the Windows Virtual Shiled for Arduino

Iam using Microsoft Lumia 535
Arduino Uno
HC-05 Bluetooth Module
Arduino IDE 1.6.8

I am possible to connect the Bluetooth with virtual shiled for Arduino after the bluetooth connection my app get hang or crash unable to control the Arduino with that and I am unable to work with that even simple hello world text  I am unable seen in the virtual shield App

JimGaleForce, 2016-03-23:
kumareshmsp221 : Both the Arduino library and the Universal app have been updated in their stores. Please retry, and in your code, specify the correct baudrate in the sample/code with shield.begin(34800) or shield.begin(9600)

balintmaci, 2016-05-22:
Hi, I tried it with a 
nano, duemilanova, mega
hc 05, hc 06 bluetooth chips. latest library (1.2), latest app and the same thing happens. as soon as it connects the app freezes. I tried multiple baud rates including 9600. Can you please help?

JimGaleForce, 2016-06-07:
can you share the version of Win10 you are using along with the Arduino IDE version?

balintmaci, 2016-06-07:
replied in the other thread.

IoTGirl, 2017-06-20:
Closing as stale - Looks like the topic moved to a different thread and there has been no movement on this one for more than a year.

closed 2017-06-20 by IoTGirl