ms-iot/virtual-shields-arduino #31: Access Virtual Shield from Raspberry Pi

pdr0663, 2016-05-14:
I see from:

#if defined(ARDUINO) && ARDUINO > 100
#elif defined(_WINRT_DLL)
#endif

that there is an alternative to access a Windows device other than from an Arduino. I'm interested in accessing a Windows Phone for IoT applications, and would like to do so from a Raspberry Pi using Raspian Linux. My dream scenario is to do so with Javascript and Node.js. I understand that node is probably not of interest in this project, but can the necessary bits be added here to allow access from a Linux system?

Anyone interested in a node.js NPM package for the Virtual Shield? Would be great!

Paul

pdr0663, 2016-06-21:
How about it sidwarkd? Seem like a good idea?

sidwarkd, 2016-06-21:
@pdr0663 Really sorry about the delayed response. Been travelling a ton the last month. So it sounds like you're really looking for two separate things. 1) You would like access to Virtual Shields (Windows Phone interactions) from Raspbian instead of Windows IoT or Arduino and 2) A Node.js package for doing Virtual Shield interactions from javascript. If I missed something let me know. The node.js portion should really be a separate project and could either be written from the ground up or could even be written as a C extension on top of the Virtual Shield core library. Either way I agree it would be awesome but would also be a large undertaking. For the first item it would be a matter of hooking in the underlying serial protocol code that runs on Raspbian.

So right now on both Arduino and Windows 10 IoT Core the communication for Virtual Shields is done via the serial port lines (TX and RX) and then you can attach bluetooth or WiFi shields to those lines to enable BT and WiFi comm with the phone. Would that be an acceptable scenario for you on Raspbian or would you be looking for native bluetooth or WiFi access without using the RX and TX lines?

pdr0663, 2016-06-26:
Thanks for your detailed response, it is much appreciated.

Yes your summation of my request is accurate. I currently use johnny-five (see www.johnny-five.io) which allows an arduino with Firmata software to be connected by USB to the pi. Access to the IO resources of the Arduino is then child's play.

I would expect that a BT dongle or the build-in BT of the Pi3 would be supported with Linux drivers, so that would be the most rational hardware to achieve the result, and preferred. Similarly, there is good native support for WiFi.

I'm not sure of the pi's capabilities regarding BlueTooth over it's IO pins under Linux. However, a  BT module on IO pins is fairly trivial hardware-wise, so would be OK. 

If the C++ side could be worked-out for Raspbian, I could try to get the node.js community interested in an NPM package.

Thanks again for your reply!

IoTGirl, 2017-06-20:
Closing Stale - no post from originator or owner in more than a year

closed 2017-06-20 by IoTGirl