ms-iot/virtual-shields-arduino #32: Arduino SAM Board Support (Not SAMD)

rmcolbert, 2016-05-20:
Attempting to compile the example sketchs for the Arduino Due board result in compiler errors about re-declaring Bool.  Output from the compiler below:

`In file included from C:\Users\xxx\Documents\Arduino\libraries\Windows_Virtual_Shields_for_Arduino\src/Attr.h:28:0,
                 from C:\Users\xxx\Documents\Arduino\libraries\Windows_Virtual_Shields_for_Arduino\src/VirtualShield.h:40,
                 from C:\Users\xxx\Documents\Arduino\libraries\Windows_Virtual_Shields_for_Arduino\examples\HelloWorld\HelloWorld.ino:27:

C:\Users\xxx\Documents\Arduino\libraries\Windows_Virtual_Shields_for_Arduino\src/SensorModels.h:47:9: error: 'Bool' redeclared as different kind of symbol
  Bool = 7,
         ^

In file included from C:\Users\xxx\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.8\system/libsam/chip.h:66:0,
                 from C:\Users\xxx\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.8\cores\arduino/Arduino.h:42,
                 from sketch\HelloWorld.ino.cpp:1:

C:\Users\xxx\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.8\system/libsam/include/USB_device.h:74:33: error: previous declaration of 'typedef unsigned char Bool'
 typedef unsigned char           Bool; //!< Boolean.
                                 ^

In file included from C:\Users\xxx\Documents\Arduino\libraries\Windows_Virtual_Shields_for_Arduino\src/Attr.h:28:0,
                 from C:\Users\xxx\Documents\Arduino\libraries\Windows_Virtual_Shields_for_Arduino\src/VirtualShield.h:40,
                 from C:\Users\xxx\Documents\Arduino\libraries\Windows_Virtual_Shields_for_Arduino\examples\HelloWorld\HelloWorld.ino:27:

C:\Users\xxx\Documents\Arduino\libraries\Windows_Virtual_Shields_for_Arduino\src/SensorModels.h: In constructor 'EPtr::EPtr(const char*, bool)':

C:\Users\xxx\Documents\Arduino\libraries\Windows_Virtual_Shields_for_Arduino\src/SensorModels.h:214:53: error: expected primary-expression before ')' token
     EPtr(const char* key, bool value) : ptrType(Bool), key(key), boolValue(value) {}
                                                     ^

exit status 1
Error compiling for board Arduino Due (Programming Port).`

zfields, 2016-06-07:
Thank you for posting! We are looking into the best way to solve this problem. Thanks!

saraclay, 2018-08-10:
Closing this out as it's been a while. Feel free to file an issue on ms-iot/content as we are more active there.

closed 2018-08-10 by saraclay