ms-iot/virtual-shields-arduino #33: App feezing

balintmaci, 2016-05-24:
Hi!
I tried it with an arduino nano, duemilanova, mega
hc 05, hc 06 bluetooth chips
latest library (1.2)
latest app
the same thing happens. as soon as the app connects it freezes. I tried multiple baud rates including 9600. 
also I tried it on my pc as well as my lumia 950
Do you know what might cause this issue?
Thanks

omikcity, 2016-06-01:
Hi,
I am using Arduino Mega board and HC-05 Bluetooth module.
I want to use virtual shield but the problem is as I connect my lumia 930 with HC-05 bluetooth module which is connected to arduino mega, my app freezes and crash after some time.
I am not able to use virtual shields, I googled a lot about this issue but did not find any solution.
Kindly help me.

JimGaleForce, 2016-06-07:
what version of Windows 10 are you running on your phone(s)?

balintmaci, 2016-06-07:
Right now both my pc and phone is on slow ring so that's 14342 for pc and 14295 for phone, but had tried this app months ago and I had the same issue.
What's weird is I have no problem communicating with the arduino using the Bluetooth Serial Terminal app, so something must be wrong with the app. Any idea why it freezes?

LycheeYum, 2016-08-31:
I'm am experiencing this as well.  Both PC and phone and connect through terminal apps.  However, the app freezes when a bluetooth connection is made.  I've tried all the baud settings and still no joy.

Windows 10 AU PC
950XL Windows 10 mobile AU 
HC-06 1.8
Virtualshield 1.2.0
JSon 5.6.6

Thank you

omikcity, 2016-08-31:
I have figured out it. The solution is, download the source code of this app and deploy it to your phone/pc through visual studio 2015 instead of downloading through store.This solved my issue. 

Date: Tue, 30 Aug 2016 20:35:54 -0700

balintmaci, 2016-08-31:
That is amazing

LycheeYum, 2016-09-01:
Thank you for the reply.  I can confirm that deploying the solution via Visual Studio works on both Windows 10 and Windows 10 Mobile.

ooeygui, 2016-12-07:
Looks like the app needs an update.

IoTGirl, 2017-06-19:
Seems to be solved - closing this issue.

closed 2017-06-19 by IoTGirl

Frostsovereign, 2017-11-22:
Hello , can i plz have the appx to deploy to my windows phone , i am having the same issue , thnx.

omikcity, 2017-11-22:
Hi Frostsovereign,

I’ll send you the appx files once I’ve found It on my system.

Thaks.

Frostsovereign, 2017-11-22:
thank you i am waiting :)

Frostsovereign, 2017-11-22:
Hello , I am still waiting for the file, any good news, have you found it yet?

Frostsovereign, 2017-11-24:
Can anyone please help me , i need the appx to deploy

Frostsovereign, 2017-11-26:
hello, any good news ? i have been waiting for many days :(