ms-iot/virtual-shields-arduino #35: Virtual shield windows application freezes as I connect with bluetooth HC-05

omikcity, 2016-06-01:
Hi,
I am using Arduino Mega board and HC-05 Bluetooth module.
I want to use virtual shield but the problem is as I connect my lumia 930 with HC-05 bluetooth module which is connected to arduino mega, my app freezes and crash after some time.
I am not able to use virtual shields, I googled a lot about this issue but did not find any solution.
Kindly help me.

LycheeYum, 2016-08-31:
Experiencing the same.  Any assistance would be appreciated.

omikcity, 2016-08-31:
Just install the app by downloading source code from git and deploy to your phone/pc using visual studio 2015. If you download app from the store, app will freeze.
Date: Tue, 30 Aug 2016 20:27:30 -0700

LycheeYum, 2016-09-01:
Thank you for the reply.  I can confirm that deploying the solution via Visual Studio works on both Windows 10 and Windows 10 Mobile.

ooeygui, 2016-12-07:
I'm going to duplicate this to https://github.com/ms-iot/virtual-shields-arduino/issues/33.

closed 2016-12-07 by ooeygui