ms-iot/virtual-shields-arduino #36: Can't pair on USB.

Brainer189, 2016-07-06:
I've downloaded the app, installed the libraries, but the app says no paired devices found.

ooeygui, 2016-12-07:
Did you install the firmware on the Arduino? 
What bluetooth module are you using?

ooeygui, 2016-12-07:
Sorry for the delay in responding.
Which arduino are you using?

Brainer189, 2017-01-17:
I have used an Arduino Uno. I havent used a bluetooth module. It doesnt
works without it?

IoTGirl, 2017-01-24:
Hi Brainer,
There is a discussion on a similar topic on the forums. Is this what you are attempting? [https://social.msdn.microsoft.com/Forums/en-US/1db42e25-79bf-4f50-a899-4948ce266bc1/windows-virtual-shield-connection-via-usb-does-not-work?forum=WindowsIoT](https://social.msdn.microsoft.com/Forums/en-US/1db42e25-79bf-4f50-a899-4948ce266bc1/windows-virtual-shield-connection-via-usb-does-not-work?forum=WindowsIoT)
Sincerely,
IoTGirl

IoTGirl, 2017-06-21:
Closing as stale - no follow up for months.  Others have had success downloading the source and using their own copy off the connection app rather than the one from the store.

closed 2017-06-21 by IoTGirl