ms-iot/virtual-shields-arduino #38: does not compile with teensy

tonton81, 2017-01-02:
i can compile it fine with mega2560 but it wont compile on my teensy 3.2 or 3.5 which uses arduino IDE 1.8
with teensyduino 1.34 support.

tonton81, 2017-01-02:
i would like to start using it with my surface book
thanks

IoTGirl, 2017-01-24:
Hi Tonton81,
That is correct Virtual shields is for Arduino and is not expected to support Teensy. As stated on their site they expect many libraries to work but not all. See [https://www.pjrc.com/teensy/td_libs.html](https://www.pjrc.com/teensy/td_libs.html)
Sincerely,
IoTGirl

closed 2017-01-24 by IoTGirl

tonton81, 2017-01-25:
well that was discouraging considering they are still using the arduino core

IoTGirl, 2017-01-31:
It seems that they may be a clone but are very honest that this is a different platform and that they don't support all Arduino capabilities.  See the differences section on their site [https://www.pjrc.com/teensy/teensyduino.html](https://www.pjrc.com/teensy/teensyduino.html)